JuliaRobotics/DistributedFactorGraphs.jl #711: Tag v0.11.1

GearsAD, 2020-12-11:
@JuliaRegistrator register branch=release/v0.11

JuliaRegistrator, 2020-12-11:
Registration pull request created: [JuliaRegistries/General/26214](https://github.com/JuliaRegistries/General/pull/26214)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.11.1 -m "<description of version>" ca3ee9f7ac436a44f7753a7ef781b38edd17640c
git push origin v0.11.1
```

dehann, 2020-12-11:
Hi @GearsAD , not sure if you know -- just in case.  The workflow file for TagBot will trigger new tag on closed issues (if there is anything to tag)

closed 2020-12-11 by dehann

dehann, 2020-12-13:
@GearsAD , okay think that is all of them:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/milestone/64?closed=1