JuliaRobotics/DistributedFactorGraphs.jl #727: Upstream BiDictMap to DataStructures.jl?

dehann, 2021-01-17:
This is most useful when trying to make LightGraphs SimpleGraph using Symbol labels, so perhaps `BiDictMap` should be a feature of their upstream DataStructures.jl so that we can just use downstream in more than one repo?

Affie, 2021-01-19:
Way back when I wrote this there wasn't another solution, this was basically  what i found:
https://discourse.julialang.org/t/invertable-dict-structure/7225

With a quick search it looks like there might be some options available now (I haven't actually looked in detail at one of them):
https://discourse.julialang.org/t/datastructure-for-two-way-map/52735
https://github.com/scheinerman/Bijections.jl
https://github.com/rikhuijzer/BidirectionalMaps.jl
https://github.com/JuliaData/CategoricalArrays.jl

dehann, 2021-01-19:
Since you wrote Johan (don't want to steal your thunder), perhaps good to just make it known on those threads ... perhaps even better to just do a PR with test to DataStructures.jl and then also xref with each of the threads you point out.  Someone should just start a BiMap and then the world will start hacking at it.  I looked at DataStructures.jl the other day and did not see a BiMap/BiDict hence this issue.  Would you be up for something like that?