JuliaRobotics/DistributedFactorGraphs.jl #729: Transition CI from Travis to Github Actions

dehann, 2021-01-19:
Part of the general migration from Travis.  Also requires fix for recent Neo4j install failure on Travis.  Jim, going to assign since we spoke, hope that's okay?  Thanks!

jim-hill-r, 2021-01-20:
Is the intent of this branch regex to do CI on tagged branches? /^v\d+\.\d+(\.\d+)?([-+]\S*)?$/ Github actions has the notion of just doing ci based on when a tagging happens. Can we just use that?

dehann, 2021-01-20:
Thanks for taking a look!  We have been testing DFG against `master`, `release/v0.x`, and one integration test with downstream package IIF, branch:`dfg_integration_test`, https://github.com/JuliaRobotics/IncrementalInference.jl/tree/upstream/dfg_integration_test

Not sure about the regex branch, @Affie , see you added that as test option -- what is the plan/usage there?  I'm not immediately seeing where that branch is pointing either:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/a6451e3a5ac80a8ec686affb02fbbad3ab7d6c1c/.travis.yml#L25

Jim, also note 
- the `develop` branch comes and goes from time to time.
- Testing is segmented to also test with or without CloudGraphDFG via ENV variable `SKIP_CGDFG_TESTS` (Neo4j install not required if skipped)

I'd say the long term usage for CI on DFG is that (and IIF integration test against branch `dfg_integration_test` should be kept for all):
- PRs should run when created/updated,  
- `master` when PR merged or cron once a week-ish, 
- `release/v0.X` should run once a week-ish

Full stack test is important so `SKIP_CGDFG_TESTS` not really required, was just convenient when different parts were breaking -- can restore later if easier.

I'd say time limit to like 60mins max effort, and get up whatever you can in that time.  We can then take the weeks to figure out how to migrate everything.  Travis should still be available for a little while.

Affie, 2021-01-20:
> Is the intent of this branch regex to do CI on tagged branches? /^v\d+.\d+(.\d+)?([-+]\S*)?$/ Github actions has the notion of just doing ci based on when a tagging happens. Can we just use that?

Hi @jim-hill-r, that was to build on tags. It was for building the tags on documentation:
[image]

PR #730 (2021-06-29): Converted travis to github ci
Fixes #729

Still needs
- [ ] Testing in github to validate it works
- [x] Updated badging
closed 2021-06-29 by dehann