JuliaRobotics/DistributedFactorGraphs.jl #791: Why DefaultOrthonormalBasis()?

Affie, 2021-07-29:
@dehann, do you have a specific reason to use `DefaultOrthonormalBasis()` in `getPoint` and `getCoordinate`?

https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/f002086fced0265be0a55938e086aeb6942a5d63/src/services/DFGVariable.jl#L162-L167

I just want to make sure you intended to use it as I was expecting `vee`/`hat` and it caused some problems. 

Note that `DefaultOrthonormalBasis` is not the basis we normally use:

```julia

julia> v = [0; 0; -pi];

julia> M = getManifold(Pose2);

julia> p0 = getPointIdentity(Pose2);

julia> X = get_vector(M, p0, v, DefaultOrthonormalBasis())
ProductRepr with 2 submanifold components:
 Component 1 =
  2-element Vector{Float64}:
   0.0
   0.0
 Component 2 =
  2×2 Matrix{Float64}:
    0.0      2.22144
   -2.22144  0.0

julia> X = hat(M, p0, v)
ProductRepr with 2 submanifold components:
 Component 1 =
  2-element Vector{Float64}:
   0.0
   0.0
 Component 2 =
  2×2 Matrix{Float64}:
    0.0      3.14159
   -3.14159  0.0
```

Affie, 2021-07-29:
As a test, I changed it to vee/hat and my `Pose2Point2BearingRange` tests is now closer to what I expected.

dehann, 2021-07-30:
oh, i was working from Ronny's suggestion that we are using DefaultOrthogonalBasis().  Remember that vee and hat are just wrappers for get_coordinates and get_vector in Manifolds.jl.  We can just change it to vee and hat if that works easier for you now.

Affie, 2021-07-30:
Can we then perhaps change it to `DefaultOrthogonalBasis` then with a `basis` default/keyword argument to change it to `DefaultOrthonormalBasis` if you need it somewhere?

PR #792 (2021-07-30): Change to DefaultOrthogonalBasis and add as arg
close #791
closed 2021-07-30 by dehann