JuliaRobotics/DistributedFactorGraphs.jl #797: Julia 1.6.2 testing error

sam-globus, 2021-08-15:
I didn't have time to check what's causing the issue but I'm seeing an error when testing DFG with Julia 1.6.2 (latest Docker image). It seems there's an issue with the `@defVariable` macro.

Stacktrace:
```
ERROR: LoadError: LoadError: LoadError: MethodError: no method matching var"@defVariable"(::LineNumberNode, ::Module, ::Symbol, ::Expr)
Closest candidates are:
  var"@defVariable"(::LineNumberNode, ::Module, ::Any, ::Any, ::Any) at /root/.julia/packages/DistributedFactorGraphs/l1oET/src/services/DFGVariable.jl:92
Stacktrace:
 [1] include(fname::String)
   @ Base.MainInclude ./client.jl:444
 [2] top-level scope
   @ ~/.julia/packages/DistributedFactorGraphs/krlsn/test/runtests.jl:28
 [3] include(fname::String)
   @ Base.MainInclude ./client.jl:444
 [4] top-level scope
   @ none:6
in expression starting at /root/.julia/packages/DistributedFactorGraphs/krlsn/test/testBlocks.jl:19
in expression starting at /root/.julia/packages/DistributedFactorGraphs/krlsn/test/testBlocks.jl:19
in expression starting at /root/.julia/packages/DistributedFactorGraphs/krlsn/test/runtests.jl:28
ERROR: Package DistributedFactorGraphs errored during testing
```

dehann, 2021-08-22:
This is what I'm expecting to see on testBlocks.jl Ln19:
```julia
@defVariable TestVariableType1 Euclidean(1) [0.0;]
```

Note the new code wants an `AbstractManifold` and reference point on that manifold.  This is all part of the new Manifolds.jl version of the code, Specifically IIF v0.25.  Lets just figure out first if this issue is from a versionitis thing, or if there a forward-backward compat issue, or if there actually is a bug.  I'm thinking it's the first, maybe second -- I don't think at this stage there is a bug with `@defVariable` since we have been working with it a lot recently.

------

PS, separately, we are actually moving away from `Euclidean(N)` and instead moving towards `TranslationGroup(N)`

GearsAD, 2022-01-09:
I believe this was resolved earlier and this is stale.

closed 2022-01-09 by GearsAD