JuliaRobotics/DistributedFactorGraphs.jl #816: Serialization: SolverDataDict and associated elements should not be strings in the packed type.

GearsAD, 2021-11-08:
They should be fully-fledged structures.

GearsAD, 2021-11-08:
Specifically, these should be converted to complete structures:
- solverDataDict
- ppeDict
- dataEntry
- smallData

Reference structure:
```
{
   "label":"x0",
   "dataEntry":"{}",
   "nstime":"0",
   "dataEntryType":"{}",
   "ppeDict":"{}",
   "solverDataDict":"{\"default\":{\"vecval\":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],\"dimval\":3,\"vecbw\":[0.0,0.0,0.0],\"dimbw\":3,\"BayesNetOutVertIDs\":[],\"dimIDs\":[0,1,2],\"dims\":3,\"eliminated\":false,\"BayesNetVertID\":\"_null\",\"separator\":[],\"variableType\":\"RoME.Pose2\",\"initialized\":false,\"infoPerCoord\":[0.0,0.0,0.0],\"ismargin\":false,\"dontmargin\":false,\"solveInProgress\":0,\"solvedCount\":0,\"solveKey\":\"default\"}}",
   "smallData":"{}",
   "variableType":"RoME.Pose2",
   "solvable":1,
   "tags":"[\"VARIABLE\"]",
   "timestamp":"2021-10-30T13:54:44.468-05:00",
   
```

Affie, 2024-10-21:
This is the case except for factor data

closed 2024-10-21 by Affie