JuliaRobotics/DistributedFactorGraphs.jl #828: DFG v0.18 Laundry List

dehann, 2022-01-06:
- [ ] Fix chicken egg, IIF 1424,
- [ ] Distributions and JSON,
- [ ] Base64 on packed factor data,
- [ ] LightGraphs -> Graphs maintenance,
- [ ] User / Robot session creation optimization,
- [ ] proper DFG alias and driver names,
- [ ] Should we pull out Neo4JDFG as separate repo

GearsAD, 2022-01-17:
Added a [board for this on NavAbility](https://github.com/orgs/NavAbility/projects/34) to help track it. To discuss: we should make core projects in JuliaRobotics for public visibility.

dehann, 2022-01-31:
Thanks, I just updated the Epic board.  One or two outstanding issues for DFG v0.18.0.  @GearsAD , did you want us to complete #839 for DFG v0.18.0 or defer to v0.18.1?

dehann, 2022-01-31:
Deferred #839

closed 2022-01-31 by dehann