JuliaRobotics/DistributedFactorGraphs.jl #835: AddFactor should not require the variables to exist

GearsAD, 2022-01-09:
As a user I would like to add factors before their variables exist. In CGDFG this can be done by adding phantom variables until the full variable exists.

TBD should this be added to all DFG or just CGDFG?

Affie, 2022-01-13:
Here is the relevant issue: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/510

> As a user I would like to add factors before their variables exist

I understand the cloud use case, why local? It is doable but adds a layer of complexity.

jim-hill-r, 2022-01-14:
Could we discuss this for local in an engineering talk sometime down the line. I have often wondered what production application code would look like. As I understand it, a sensor reading is effectively providing a factor, so the application code has to create a factor and the associated variables. Does it make sense to just make a function like addSensorReading(someFactorConstructor(), [someVariableConstructor(), someVariableConstructor()].. ie, just pump all the data at once? That would help me understand the need for "phantom" idea on local.

dehann, 2025-08-05:
Replaced by:
- #1167

closed 2025-08-05 by dehann