JuliaRobotics/DistributedFactorGraphs.jl #856: Tag v0.18.0

Affie, 2022-02-01:
@JuliaRegistrator register branch=release/v0.18

Affie, 2022-02-01:
@JuliaRegistrator register branch=release/v0.18

JuliaRegistrator, 2022-02-01:
Registration pull request created: [JuliaRegistries/General/53608](https://github.com/JuliaRegistries/General/pull/53608)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.18.0 -m "<description of version>" ebdd8b64f2f512f218201becd9caf5071c86e998
git push origin v0.18.0
```

dehann, 2022-02-01:
cool thanks, i also posted it in slack announcements channel

closed 2022-02-01 by Affie