JuliaRobotics/DistributedFactorGraphs.jl #878: Tag v0.18.3

GearsAD, 2022-04-13:
@JuliaRegistrator register branch=release/v0.18

JuliaRegistrator, 2022-04-13:
Registration pull request created: [JuliaRegistries/General/58431](https://github.com/JuliaRegistries/General/pull/58431)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.18.3 -m "<description of version>" 35d4307f24089b353e6f8ef952d019c152dfa23e
git push origin v0.18.3
```

dehann, 2022-04-13:
cool, registry PR has merged so, im going to close this issue to trigger local Tagbot to tag the v0.18.3 release locally.

closed 2022-04-13 by dehann