JuliaRobotics/DistributedFactorGraphs.jl #888: Neo4jDFG should support parallel sentinel creation (user/robot/session)

GearsAD, 2022-08-17:
As a user I want the sentinel creation to support parallel calls to create the sentinels. If I call addVariable in two threads for the same worker, there should not be any issue with parallel creation of sentinels.

Affie, 2022-08-17:
I’ve always thought the “create if not exist” was a hack to get it to work. Is it not easier to create a session explicitly and fail if it’s not there? 
on the other hand i don’t have a problem with this. Just want to prioritize.

GearsAD, 2022-08-17:
Fair, I figure it's more of a usability issue. As a user I don't want to explicitly create or check for a session before writing data, that's friction that we don't want to introduce.

Affie, 2022-08-19:
I think I misunderstood the issue.

GearsAD, 2022-09-14:
This has been done and will be available in the next release.

closed 2022-09-14 by GearsAD