JuliaRobotics/DistributedFactorGraphs.jl #897: tag v0.18.5

dehann, 2022-08-25:
@JuliaRegistrator register branch=release/v0.18

JuliaRegistrator, 2022-08-25:
Registration pull request created: [JuliaRegistries/General/67019](https://github.com/JuliaRegistries/General/pull/67019)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.18.5 -m "<description of version>" 10029563b04f0c5c907bd5a78cff1e02d3099855
git push origin v0.18.5
```

dehann, 2022-08-25:
@JuliaRegistator register branch=release/v0.18

dehann, 2022-08-25:
@JuliaRegistrator register branch=release/v0.18

JuliaRegistrator, 2022-08-25:
Registration pull request updated: [JuliaRegistries/General/67019](https://github.com/JuliaRegistries/General/pull/67019)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.18.5 -m "<description of version>" 3a893150bd624b3812f6f74f2e58cf26f9f2c1bc
git push origin v0.18.5
```

closed 2022-08-25 by dehann