JuliaRobotics/DistributedFactorGraphs.jl #939: DataEntry.acquisitionTime

dehann, 2023-01-25:
TAC

- add field to data entry

dehann, 2023-03-17:
xref:
- https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/985

GearsAD, 2023-03-17:
> Is that not the timestamp field? Why both?

Yep, I made it simply timestamp, that work for everyone?

The automated fields are lastUpdatedTimetstamp and createdTimestamp.

dehann, 2023-03-27:
ah, missed the last comment -- so confirming:
- `BlobEntry.timestamp` is the same as `.acquisitionTimestamp`.
- `createdTimestamp` is as the name suggests.
- `lastUpdatedTimestamp` ... if a mutation happens for something specific like assigning a `.blobId` upon server upload.

Affie, 2023-03-27:
I don't think `createdTimestamp` is intended to be set by the user but I'ts something interesting to consider in the offline use case.

closed 2024-08-31 by Affie