JuliaRobotics/DistributedFactorGraphs.jl #971: Non trivial pattern for deserializing ZoneDateTime

dehann, 2023-03-01:
This gives a parsing error:

```julia
using JSON3

struct MyType
  createdTimestamp::ZonedDateTime
end

str = "{\"createdTimestamp\":\"2020-08-11T04:05:59.82-04:00\"}"

JSON3.read(str, MyType)
```

Affie, 2023-03-02:
That is unfortunate, it used to work out of the box. The default format for parsing is "yyyy-mm-ddTHH:MM:SS.ssszzz" so it's missing the 3rd decimal on seconds.

Affie, 2023-03-02:
It looks like its related to this issue: https://github.com/JuliaTime/TimeZones.jl/issues/83#issuecomment-502243122

If you create the string with JSON3 it looks like it works fine. as it writes the third decimal place on seconds.

dehann, 2023-04-04:
Think this was fixed in DFG v0.20?  If so then we can likely close this.

closed 2023-04-19 by Affie