JuliaRobotics/DistributedFactorGraphs.jl #988: tag v0.20.1

dehann, 2023-03-13:
@JuliaRegistrator register branch=release/v0.20

JuliaRegistrator, 2023-03-13:
Registration pull request created: [JuliaRegistries/General/79465](https://github.com/JuliaRegistries/General/pull/79465)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.20.1 -m "<description of version>" d7b4fdc7917d98c09dbdb33f4459b7a2809bf92c
git push origin v0.20.1
```

Affie, 2023-04-21:
@JuliaRegistrator register branch=release/v0.20

JuliaRegistrator, 2023-04-21:
Registration pull request created: [JuliaRegistries/General/82038](https://github.com/JuliaRegistries/General/pull/82038)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.20.2 -m "<description of version>" 21804f88edbe117225526cd424fcac4360708001
git push origin v0.20.2
```

closed 2023-04-21 by Affie