JuliaRobotics/DistributedFactorGraphs.jl #1006: tag v0.21

Affie, 2023-04-16:
@JuliaRegistrator register branch=release/v0.21

JuliaRegistrator, 2023-04-16:
Registration pull request created: [JuliaRegistries/General/81689](https://github.com/JuliaRegistries/General/pull/81689)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.21.0 -m "<description of version>" 5fccbe0ecb2fe217921ee70b97e3174d3fd12b73
git push origin v0.21.0
```

Affie, 2023-04-16:
@JuliaRegistrator register branch=release/v0.21

JuliaRegistrator, 2023-04-16:
Registration pull request updated: [JuliaRegistries/General/81689](https://github.com/JuliaRegistries/General/pull/81689)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.21.0 -m "<description of version>" 6d42669706e9f6ada0d6cc4851293005107369d7
git push origin v0.21.0
```

JuliaRegistrator, 2023-04-17:
Registration pull request created: [JuliaRegistries/General/81760](https://github.com/JuliaRegistries/General/pull/81760)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.21.1 -m "<description of version>" 9b6f93b98fa2367b603c417c718bba411fc60739
git push origin v0.21.1
```

closed 2023-04-17 by Affie