JuliaRobotics/DistributedFactorGraphs.jl #1047: tag v0.23.2

dehann, 2023-09-05:
@JuliaRegistrator register branch=release/v0.23

JuliaRegistrator, 2023-09-05:
Registration pull request created: [JuliaRegistries/General/90833](https://github.com/JuliaRegistries/General/pull/90833)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.23.2 -m "<description of version>" 180f2efbcb69dbc91acad6814216e5c22f0a36d7
git push origin v0.23.2
```

dehann, 2023-09-05:
@JuliaRegistrator register branch=release/v0.23

JuliaRegistrator, 2023-09-05:
Registration pull request updated: [JuliaRegistries/General/90833](https://github.com/JuliaRegistries/General/pull/90833)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.23.2 -m "<description of version>" 180f2efbcb69dbc91acad6814216e5c22f0a36d7
git push origin v0.23.2
```

closed 2023-09-05 by dehann