JuliaRobotics/DistributedFactorGraphs.jl #1084: copyGraph is using the wrong verb

dehann, 2024-07-11:
in call with Johan on verb Nouns for how to "get a copy of graph from one machine to another"
- `function merge!(dst, srcs...; *labels=:) # returns dst, is blocking`
- `duplicate(src) = merge!(LocalDFG(), src) # is blocking (should this be a push or pull verb?)`
- `clone(src) # does not overwrite`

Affie, 2024-07-11:
I'm also considering `get`.
- Currently the `session` noun, but there exists a session structure and it will be deprecated.
    - How to distinguish between getting the container and the container plus links. 
    - getVariable gets the variable + satellites. Maybe only satellites included automatically in get. 
- save/load verbs uses noun `DFG` so `getDFG(NvaClient, sessionLabel)`
- copy uses noun `Graph` so `getGraph(NvaClient, sessionLabel)`

Affie, 2025-02-04:
We should also look into `fetch` and `post` for higher level async verbs.

Affie, 2025-06-10:
We discussed `sync` as a new verb for a higher-level copy/merge to function.