JuliaRobotics/DistributedFactorGraphs.jl #1085: Review verbs for DFG v1.0

Affie, 2024-07-11:
Verbs here: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/wiki/Standard-Definitions-of-API-via-verbs-and-set-theory
With regard to:
- Signature
- Return Type
- Error Behaviour

Verbs:
- [ ] list
- [ ] get
- [ ] add
- [ ] update
- [ ] delete

TODO add all verbs for DFG v1.0

Affie, 2024-08-28:
**`list`**
Synchronously list the labels of an existing (composite) data container/object in an intelligent way
- Synchronous
- Example nouns Variables, Factors, VariableNodeData, PPE -> Vector{Symbol} 
- Returns empty list if the container exists but empty
- Throw typed Exeption `KeyError` or custom exeption if the container does not exist.
- Intelligent way:
    - Regex filter on label
    - Filter on Solvable
    - Filter on Tags (tag in variable.tags)

dehann, 2025-06-29:
I disagree with keyword specification for what to updated.  This is taking CRUD too far.

I agree with using merge for now and keeping update as a reserved verb in the mean time.

Therefore, I would expect DFG v1 usage would generally look something like this:

```julia
v1 = getVariable(srcfg, :x1)
mergeTags(v1, [:NEW_TAG;])
setSolvable!(v1, 4)
merge(srcfg, v1)
```

Affie, 2025-07-06:
`has`
Return a boolean indicating whether the object exists by `label::Symbol`. The api is similar to get:
```julia
hasVariable(dfg, label::Symbol)
```

dehann, 2025-07-31:
> `has`, Return[s] a boolean indicating whether the object exists by `label::Symbol`. [Signatures are similar] to `get`.

Sounds good to me, note I updated the proto-definition text a bit above.