JuliaRobotics/DistributedFactorGraphs.jl #1105: All node metadata design

dehann, 2024-11-05:
TAC:
- [ ] rename last remaining smallDatas
- [ ] `JSON3.Object` fast enough for `Compute_Node_` level?
  - [ ] Delete smalldata types union
  - [ ] Allows better nesting for JSON types

Affie, 2024-11-12:
My vote is to keep the design as is with `SmallDataTypes`. Its been this way for 4 years without any issues. see https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/pull/576

Affie, 2024-12-02:
Partially addressed in #1107

Affie, 2025-02-17:
Also considered just using `Dict{Symbol, String}` or creating a new struct such as

Affie, 2025-03-21:
> SmallDataTypes good, that is equivalent to just using existing JSONable content.

Yes, idea of SmallDataTypes is to force JSONable content.

> That also allows nesting of dicts (which we want).

Can you elaborate on this please? We cannot currently nest dicts, metadata in essence is just key value pairs. The value itself can be a json serialized dict as a string. 

Do you agree on metadata as just key value pairs to help keep it small as anything bigger should be stored in a blob?

Affie, 2025-03-26:
Does metadata have CRUD?

Currently add/update/delete would have to: get, add/update/delete, set. That makes Isolation not possible, so we would have to change the structure and implement it in the backend for proper CRUD.

dehann, 2025-03-29:
> The value itself can be a json serialized dict as a string. [Can you elaborate on this please?]
> [NOPE] Only the UX of having to convert strings to variables.
> [NOPE] Using Dict{Symbol, String} might be the easiest way ...

No this this the same thing that keeps coming back.  NEVER put a JSON.write inside a value of a key=>value pair.  That is the whole point.  Pure JSON includes the idea of nested dicts.

> Do you agree on metadata as just key value pairs to help keep it small as anything bigger should be stored in a blob?

No, that is up to the user.  Keeping size of content in metadata small is not forced, that is simply convention.

> Does metadata have CRUD in the SDKs?

Yes

> Currently add/update/delete would have to: get, add/update/delete, set. That makes Isolation not possible, so we would have to change the structure and implement it in the backend for proper CRUD.

What do you mean by isolation.  Do you mean cannot guarantee atomic updates?  SDKs need CRUD, no way around that I think.

> Metadata is not meant to be used in the solver directly and can be copied into the cache if needed so performance should not be an issue.

yes, agree

Affie, 2025-03-31:
> What do you mean by isolation. Do you mean cannot guarantee atomic updates?

"Isolation" in ACID

> metadata in SDKs need CRUD, no way around that I think.

I've been using metadata without CRUD as "get" -> modify -> "set", so not needed.

Affie, 2025-03-31:
> No this this the same thing that keeps coming back. NEVER put a JSON.write inside a value of a key=>value pair. That is the whole point. Pure JSON includes the idea of nested dicts.

The current design is that smallData/metadata can't be nested, they are just key-value pairs.

Affie, 2025-03-31:
A vector of key-value pairs can avoid the whole nested JSON string issue.

dehann, 2025-04-02:
> Yes, idea of SmallDataTypes is to force JSONable content.
> The current design is that smallData/metadata can't be nested, they are just key-value pairs.

Okay, then I have been referencing SmallDataTypes incorrectly in previous comments.  Ignore reference to SmallDataTypes and lets simplify the discussion to "Nested JSON" vs "key=>value JSON only".

> "Isolation" in ACID

Okay, new name for atomic+globalized operation order.  I have been saying atomic instead of isolation.  I'll restrict my use of `atomic` to compute in a hard-struct only (i.e. lower level random access threading reads or writes).  I will use `isolate` in the higher level query/mutate context via some API. 

## From #1118 

> The options there are currently "base64 encoded JSON string" or a "JSON string"

Two risks when using "key-value JSON pattern" with random data (incl nested JSON/JSONstring artifacts in your data):
- False curly/square termination or opens `{` or `}` where data itself is interfering with JSON frames.
- Introduction of escape chars makes JSON unusable in multiple languages.  So write this today and tomorrow in a different language the JSON serde fails.

> Please elaborate why it is so bad in metadata

`.metadata` lives at the top level of the Json object.  So it is tempting to use a flat key-value layout.  While this works for small cases, it places a serious limitation on use to the point that "metadata" becomes the wrong noun.  That is why in the old days we used `smalldata` to emphasize that this is not a general purpose store.  For me, it really really sucks if every aspect of a config or parameter/config/calibration system needs to be reduced to a single flat kay-value map.  For me nesting of dicts is a vital requirement on `AnythingDFG.metadata`.

I think it is fine to have one condition on `.metadata`, and this is `jstr = json.serialize(anydfg.metadata)` should run through in any language, and that result be read in any other programming language with standard `json.deser(jstr)`.  This way we can avoid bespoke pack/unpack functions while giving user maximum freedom -- basically adopt a pass through posture on JSON.  The JSON folks figured this out, so we can just do what they do.

One layer of nesting is already included upon transmission or persistence: `json.serialize(anydfg.metadata)`, what difference does it make that metadata field itself has more nested JSON inside of it?  You mentioned database limitations, so I went reading...
- Neo4j, https://community.neo4j.com/t/can-i-add-jsonobject-as-a-value-to-a-property-in-a-node/11285
- DGraph, https://discuss.hypermode.com/t/how-to-store-read-deeply-nested-json-objects/9160 

Think at this point we should talk more on the phone about this, but if Neo4j does not support nested JSON then it is starting to become a pretty poor choice.  If that forces us to into short-term-long-term tandem plans then we should consider it.

> Packing metadata as a vector of key-value pairs can avoid the whole nested JSON string issue. Dict{Symbol, String} would be the best option then and CRUD can then be supported. Example Json would be: [{"key": "a", "value": "1"}, {"key": "b", "value": "one"}]

maybe as a workaround only, but feels like that would result in lots of wasted effort overall.

----

Side question:  this sounds like `Update` from CRUD to me: 
> I've been using metadata without CRUD as "get" -> modify -> "set", so not needed.

do you mean CRUD at driver/DB level vs CRUD at SDK level?  I've been talking at SDK level.

dehann, 2025-04-02:
Just copying link from duplicate 1118  - This should be just a JSON dict, not a JSON3.write ::String:
- https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/530653245cdc406feec7b88b4608dcefc5e2f47c/src/entities/DFGFactor.jl#L161

Affie, 2025-04-02:
> ... While this works for small cases, it places a serious limitation on use to the point that "metadata" becomes the wrong noun. That is why in the old days we used smalldata to emphasize that this is not a general purpose store.

This is currently still the case and smalldata == metadata as we should absolutely not be storing blobs in metadata for performance reasons. The definition from the dictionary is still valid on "store a couple of bytes".

> Smalldata: Pollutes the graph, but a convenient way to store a couple of bytes directly in the graph. Use this with caution and keep it small and simple

Affie, 2025-04-02:
> Side question: this sounds like Update from CRUD to me:
    >  I've been using metadata without CRUD as "get" -> modify -> "set", so not needed.
do you mean CRUD at driver/DB level vs CRUD at SDK level? I've been talking at SDK level.

I don't make a distinction between SDK and db, as a user I expect update to be ACID everywhere. DFG currently has update, but it is only valid in GraphsDFG as a reference to a variable is returned on `getVariable`.

In case it's not clear: 
- CRUD on metadata refers to CRUD on the individual keys, eg. `deleteVariableMetadata!(dfg, variableLabel, key)` deletes only the key-value pair and not the entire metadata object.
- The problem with using "get" -> modify -> "set" for updates in the SDK is that, in concurrent scenarios, the first transaction's changes can be overwritten by the last and it will be hidden from the user.

Affie, 2025-04-08:
Decisions from call on Apr 8 2025:
- There are no custom pack unpack functions, therefore only JSON'able types are supported.
- We support CRUD on metadata. Consequence:
    - key-value pairs (We want to stay away from complex custom resolvers for CRUD (and ACID))
    - Nested json double stringified value (escaped quotes `"`)

Affie, 2025-04-11:
Graphql has 2 options: 
- treat the key-value as a scalar - requires custom resolvers
- create a key-value type - this looks like the preferred solution

With the decision to support CRUD, the julia options giving the current limitations:
- Keep `SmallDataTypes` 
   - In julia
     ```julia
       :a => 1
       :b => "two"
    - json (no escaped quotes)
      ```json
        [
            {
                "key": "a",
                "value": 1
            },
            {
                "key": "b",
                "value": "two"
            }
        ]
    - nesting manually as json string
      ```json
          {
              "key": "g",
              "value": "{\"two\":2,\"one\":\"one\"}"
          }
- New wrapper type over `Any` (or complex nested)
   - Serialize wrapper as JSON string (note value is always a string): 
     ```json
        [
            {
                "key": "a",
                "value": "1"
            },
            {
                "key": "b",
                "value": "\"two\""
            }
        ]
    - Fully flexible nesting in DFG, but complex beyond key-value in SDK.
 

To support CRUD, key-value is the best option with current db. With value either:
- a JSON string or 
- `SmallDataTypes`

dehann, 2025-04-16:
HI @Affie , where was the decision captured on which way we are going here?  Was it `Dict{Symbol,String}` or `Dict{Symbol, SmallDataTypes}`?

dehann, 2025-04-16:
Hi @Affie, which option are we going with?  
- [ ] `Dict{Symbol, SmallDataTypes}` which is JSONed once as part of the node serialization, or 
- [ ] `Dict{Symbol,String}` where each key-value pair is JSONed twice (once to insert the values in dict, and once at node serialization).

Affie, 2025-04-16:
Hi @dehann, I wanted to get your input on which you prefer as well before we proceed, I'm still leaning towards `Dict{Symbol, SmallDataTypes}` as it is now because it's not JSONed twice in most cases. It will still currently need to be JSONed twice for nested structures (as a String in SmallDataTypes).

dehann, 2025-04-17:
Okay so decision is we work towards `node.metadata::Dict{Symbol, SmallDataTypes}` as fast we can.

----

Let's separate the time pressures by using blobless entries for CRUD in the meantime? (see other discussion)

Affie, 2025-04-17:
> Shall we separate the time pressures by using blobless entries for CRUD in the meantime? (see other discussion E&T 187)

Crud on variable metadata is already supported in dfg, the rest should follow over the next few minor releases to standardize to only one design. In the mean time we can use blobless entries as you suggest or the get-modify-set pattern

Affie, 2025-09-25:
The following will only work in DFG with in-memory types (mutated by reference)
```julia
addMetada(getBlobentry(getVariable(fg, vl), bel), metadatalabel)
```
and therefore, the full verbose signature is needed in general:
```julia
addVariableBlobentryMetadata(fg, vl, bel, metadatalabel)
```

dehann, 2025-09-25:
Introduction of new phantom Noun `Bloblet` for purpose of high level API that wraps blobless blobentry for storage of small data.  This is a more formal counterpart to the informal metadata field.  Convenience phantom noun Bloblet (high level API) will use blobless Blobentry (as low level key=>value) to achieve non-poweruser CRUD/ACID for "bloblet" (aka small data).  Note the following bloblet mechanizations:
- Somewhere is a node where the non-poweruser wants to do CRUD on various key=> value bloblets,
- The bloblets are stored via blobless blobentries on that node,
- the blobentry indicates it's bloblessness via `blobId=0000000-0000-0000-0000-000000000000`,
- the key is stored in the `blobentry.label`
- the value is stored in the `blobentry.metadata`
- a type boosting is possible by using `blobentry.mimetype` to overwrite the fallback metadata type `::Dict{Symbol,String}`,
  - e.g. during development someone wants to store a yaml config via the `bloblet` convenience, then the raw yaml can be stored in the blobless entry, as long as the mimetype specifies the correct type `.mimetype=application/yaml`

Affie, 2025-10-17:
I'm sure we discussed the noun `Metadata` somewhere but can't find it now. 

I recall asking what the plural form of `Metadata` was (equivalent to `getBlobentries`), @dehann , do you remember where this was to link it here and what the plural form was.

dehann, 2025-10-29:
All sounds good, go `Bloblet` as a dedicated type.

Here are my standing assumptions at this point:
- I'm assuming metadata type for all occurrences in the various DFG types remains a `::Dict{String, String}` (the long discussion above is still valid for real metadata).  That would make it similar to say S3's design (there is a metadata tab somewhere on the page for every S3 blob, but usually stuff you'd find as part of header fields in http requests to S3).  
- Metadata is data about the data (as you described above).  
  - e.g. someone wants to store details about compression algorithms, or extra http headers etc, that can all go into metadata.  Maybe more formally, things you'd expect to find in http headers are standard metadata: https://en.wikipedia.org/wiki/List_of_HTTP_header_fields
  - Use of Bloblet will win out with new users since the API for `Bloblet`s just work better when handling small data.  So even if someone tries using metadata they will eventually transition to `bloblet` or `blobentry=>blob` anyway.
    - Docs will encourage users to use `Bloblet` for small data storage (over metadata which favors more standard "metadata" eg http header example). 
    - A user that wants to store non-standard data, such as application specific sensor calibration values or whatever would then use bloblet rather than metadata.

dehann, 2025-11-11:
Constrain `.metadata` to JSONObject everywhere, does not have to support CRUD.

closed 2025-11-14 by Affie