JuliaRobotics/DistributedFactorGraphs.jl #1110: Replace Data noun with save! and load verbs

Affie, 2024-12-15:
`Data` noun to be replaced by `save!` and `load`, i.e. replace `addData` with `save!`.
As an example, we can have `saveBlob[Variable]!`, `saveBlob[Graph]!`, `saveBlob[Model]!`, `saveBlob[Agent]!`
Then also convenience functions such as `saveDFG!`, `saveFeatures!` etc.

dehann, 2025-09-16:
Examples:
- save or load images to blob

dehann, 2025-09-16:
so what about data levels, why not have for data lavel 0 something like (`Node{} is Skeleton{}`)?

dehann, 2025-09-16:
I like the idea of reducing the full verbose function name by replacement:
```
packBlobMime_addBlob!_addAgentBlobentry!
```

Reduces like algebra via:
- `save` = `pack` + `add` + `add`
- `Image` = `Blob` + `Mime` + `Blobentry`
- `Agent` = `Agent`

So it seems natural to have the reduced signature be:
```julia
saveAgentImage(::DFG, aglb, imlb, ...)
saveImageAgent(::DFG, aglb, imlb, ...)
```

I still don't think the underscores are necessary, since the user is just asking DFG/SDK/Accelerator to do thing.  Knowing high vs low lavel APII is more an internal thing to me.

Affie, 2025-09-16:
saveDFG likely only makes sense to save to a Graph or a Model (maybe Agent) and not variables and factors.

dehann, 2025-09-16:
Regarding the underscore `_`:

Affie realized (just scribing) the problem with earlier verbNounNoun reversed aliases such as `addTagAgent` vs `addAgentTag` will result in a lot of confusion, and is different from the `saveAgent` vs `saveImage` issue.

dehann, 2025-09-16:
Can we make a definition for underscore `_` in `verbNoun` signatures:
- the verb is acting on nodes separated by more than one edge in the greater DFG/NVA graph

dehann, 2025-09-16:
The underscore is more important for `delete`, e.g. 
- `deleteImage_Variable(dfg, vrlb, imlb)`

Affie, 2025-09-16:
Until we know better, we are going with `delete` because the underscore makes it clear that something else is happening -- in this case the Blob and Blobentry is deleted.

Affie, 2025-09-17:
First look at the basic signatures gives:
```julia
function loadBlob_Variable(
    dfg::AbstractDFG,
    variable_label::Symbol,
    entry_label::Symbol
)::Blob

function saveBlob_Variable!(
    dfg::AbstractDFG,
    variable_label::Symbol,
    blob::Vector{UInt8},
    entry::Blobentry,
)::Blobentry

function deleteBlob_Variable!(
    dfg::AbstractDFG,
    variable_label::Symbol,
    entry_label::Symbol,
)::Int #(numer of nodes deleted)

# other nodes example
function saveBlob_Agent!(
    dfg::AbstractDFG,
    blob::Vector{UInt8},
    entry::Blobentry,
)

function saveBlob_Graph!(
    dfg::AbstractDFG,
    blob::Vector{UInt8},
    entry::Blobentry,
)
```

Extra examples/notes on the shorthand versions:
```julia
function saveBlob!(
    dfg::AbstractDFG,#default save to graph, also Graph
    blob::Vector{UInt8},
    entry::Blobentry,
)
# Not distributed
function saveBlob!(
    agent::Agent, # getAgent(dfg)
    blob::Vector{UInt8},
    entry::Blobentry,
)
# similar in DFG and SDK
function saveBlob!(
    dfg::NavAbilityClient,
    agent::Agent,
    blob::Vector{UInt8},
    entry::Blobentry,
)
function saveBlob!(
    dfg::AbstractDFG,
    agent::Agent,
    blob::Vector{UInt8},
    entry::Blobentry,
)
function saveBlob!(
    dfg::AbstractDFG,
    variable_label::Symbol,# what about factor label
    blob::Vector{UInt8},
    entry::Blobentry,
)

#delete does not work with shorthand as deleteBlob! already exists and has a different meaning
```

Shorthand looks like only option C will be clean.

dehann, 2025-09-25:
I take it the shorthand versions are public but not exported (i.e. for powerusers)?

closed 2025-10-03 by Affie