JuliaRobotics/DistributedFactorGraphs.jl #1111: confirming signature convention for getLabel{Graph/Agent/...}

dehann, 2025-01-18:


Affie, 2025-01-22:
From discussion we decided to follow the data structure (hierarchy) when it comes to verb-noun-noun combinations ordering and alias the reverse order as well.

For example:
```julia
a = getAgent(fg)
l = getLabel(a)
# as 
l = getAgentLabel(fg)
# and can be aliased as
l = getLabelAgent(fg)
```
and
```julia
v = getVariable(fg, vlbl, ...)
d = getSolverData(v, ...)
# as 
d = getVariableSolverData(fg, vlbl, ...)
```

dehann, 2025-02-11:
Yes, for Julia universe -- I agree and just annotated with `::Type`s for clarity:
```
a = getAgent(fg::Factorgraph)
l = getLabel(a::Agent)
# as 
l = getAgentLabel(fg::FactorGraph)
# and can be aliased as
l = getLabelAgent(fg::FactorGraph)
```

closed 2025-02-13 by Affie