JuliaRobotics/DistributedFactorGraphs.jl #1114: Use verb `retrieve` for `find` + `get` verb combination.

Affie, 2025-01-31:


dehann, 2025-01-31:
Conditional +1 if good definition for list vs filter vs find?

> retrieve = (find + get)

Sounds like both list and find do the same thing?

Knee jerk reaction:

find = list + filter

Does that mean we should change DFG.list_ to not have any filters, and rather change to findVariables(dfg, tags=[:Disco;])?

dehann, 2025-01-31:
Also in bash list is not filter:

$ ls | grep -i "disco"

Affie, 2025-02-04:
> Sounds like both list and find do the same thing?

If we take inspiration from julia:
- `list` will be equivalent to `findall`
- we don't currently have `findfirst` (other than `ls()[1]`), but do have `get...First`, I think a verb may help 

> find = list + filter

That sound right to me.

> Also in bash list is not filter

I think it's important to be able to combine `list` and `get`(plural) with filters for performance.

Also see first edit on first comment

Affie, 2025-02-04:
> good definition for list vs filter vs find?

For `filter` I've just being using julia and see it as acting on a vector (ie. `Vector{Symbol}`).
`list` returns a vector of symbols from a graph for example.
`findall` = `list` + `filter`

dehann, 2025-04-01:
call on 2 April.

Affie, 2025-04-02:
- `list` see https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/1085#issuecomment-2315840695
- `filter` no a verb in DFG v1

Affie, 2025-04-02:
`find`
- in julia `findall` and `findfirst` in julia returns the index or key and is similar to `list` with filters
- at this stage no need for find in DFG v1

Affie, 2025-04-02:
`getfirst` 
- as new verb that combines `list`[1]  with `get`
- remember this is unordered
- this is a tool to help with index collisions.

Affie, 2025-08-06:
Just for completeness after you mentioned `sort`, we can include `sort` in the output to help. I'm still not convinced it's a gremlin free pattern and should not be included in DFGv1.0, but let's see if someone requests it.
Keeping for future reference:
```julia

function getfirstBlobentry(
    v::AbstractGraphVariable;
    labelFilter::Union{Nothing, Function} = nothing,
    blobIdFilter::Union{Nothing, Function} = nothing,
    sortby::Function = getLabel,
    sortlt::Function = natural_lt
)
    entries = getBlobentries(v; labelFilter, blobIdFilter)
    if isempty(entries)
        return nothing
    else
        return sort(entries; by = sortby, lt = sortlt)[1]
    end
end
```