JuliaRobotics/DistributedFactorGraphs.jl #1118: Dont nest JSON strings in pre JSON dicts (FactorDFG)

dehann, 2025-02-24:
This should be just a JSON dict, not a .write string:
- https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/530653245cdc406feec7b88b4608dcefc5e2f47c/src/entities/DFGFactor.jl#L161

Affie, 2025-02-25:
I do not agree because we don't use abstract types for type safety. 
However, factor serialization is definitely something that has to change for v1.
The 2 options as the structures currently stand are to:
- Use String (JSONObject) -  as it is now, with a string in julia.
- use PackedFunctionNodeData - but that is not a concrete type so not a good idea

My preferred solution is to change the factor structure #992

dehann, 2025-03-26:
As a general philosophy, the following is bad and we should not have this anywhere:
```
JSON.write({"label": JSON.write("namething")})
```

but there is one know exception, 
- `factor.data` #992 

And there is one existing decision on metadata:
- bad, move away from, stop doing, metadata as b64.encode(JSON.write({"meta": "my_meta"}))
- good, move towards for metadata on all nodes, just JSONObject (same as DFG.SmallDataTypes)

Affie, 2025-03-31:
> As a general philosophy, the following is bad ...

Please elaborate why it is so bad in metadata and FactorDFG.data? I don't see a way around it.

The options there are currently  "base64 encoded JSON string" or a "JSON string"

I think there will be 2 "JSON3.write(...) strings":
- The new observation function field in FactorDFG (replaces factor.data https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/1118#issue-2876271498)
- metadata pending #1105 (should discuss there)

I think we can probably close this and only discuss in the issues #1105 and #992 to avoid duplication.

dehann, 2025-04-02:
> I think we can probably close this and only discuss in the issues https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/1105 and https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/992 to avoid duplication.

okay, i was able to split the responses up, closing this.

closed 2025-04-02 by dehann