JuliaRobotics/DistributedFactorGraphs.jl #1119: Rename FactorDFG.data / fnctype becomes...

dehann, 2025-02-26:
@Affie and I had a long frustrating conversation to confirm what we already know:  That factor data fields need to be updated for DFG 1 milestone.

Context for larger work beyond this issue:
1. [This issue] Add new field to `FactorDFG.observFnc`, and replace associated old accessors such as `@deprecate getObservationFunction(::FactorDFG) getFactorFunction(::FactorCompute)`.
1. Move all other FunctionNodeData  fields to FactorDFG - #992 
1. Update CCW construction when building FactorCompute - #992 

TAC
- [ ] add new field to `FactorDFG.observFnc` (side by side with old `.data / .fnctype`)
  - Also see performance discussion (or Summary level) `FactorDFG.observFnc_type::String` #992
- [ ] implement `getObservationFunction(::FactorDFG)`
- [ ] deprecate `getFactorType` use `getObservationFunction( DFG or Compute )`

dehann, 2025-03-27:
Hi @Affie , i think we discussed a lot of this earlier today.  I'm going to look into if SDK.rs has issues with JSON (or at least how to move that forward, but after that we may need to reconnect on what we want to do with this task).

Affie, 2025-03-31:
> after that we may need to reconnect on what we want to do with this task

I only know of 2 outstanding design aspects
1. https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/1119#issuecomment-2729888859
2. Long-term plans for the FactorDFG observation function field type. My current best guess is that it will remain as a JSON serialized string.

dehann, 2025-04-02:
> I'm leaning towards only Observation as noun.

Yes, sounds good,  So there will be `observation` and `state` fields in factor.

dehann, 2025-04-25:
Unassigning from me and moved to Triage -- I finished the spike on JSONCRUD for SDK.rs and we can work around graphql_client hard-types with new callback approach for splicing in rust's `serde_json::{Map,Value}`.  So think this is a go.

dehann, 2025-04-25:
Hi @Affie think you are busy with these tasks anyway?

Affie, 2025-04-25:
Yes I think it is easiest to do the rafactor as part of #1127

Affie, 2025-05-04:
> should we parameterize FactorDFG{T}, whats going on in SDK.rs?

I'm leaning towards keeping FactorDFG as is but I'm open to changing it if there are a good reason.
We have a "factor type" field for T and non parameterized makes usage and serialization simpler.

dehann, 2025-06-19:
these last name changes are fine by me

PR #1127 (2025-06-18): Refactoring Factors: data -> state, observation and cache
Refactor factorData::GenericFunctionNodeData to state, observation and cache from #992 design.

PR should do:
- [x] partially address (FactorState) #1121
- [x] close #992 
- [x] close #1119
- [x] #1077  
- [x] update news
- [x] update docs

----

xref:
- related on Variables: #565