JuliaRobotics/DistributedFactorGraphs.jl #1121: Standardize solver data noun for variables and factors

Affie, 2025-03-17:
Currently we have the following nouns referring to solver data:

`VariableNodeData` - Struct
`PackedVariableNodeData` - Struct
`GenericFunctionNodeData` - Struct
`FunctionNodeData` = `GenericFunctionNodeData`
`data` - field in FactorDFG
`solverData` - field in FactorCompute
`solverDataDict` - field in VariableCompute
`solverData` - field in VariableDFG
`FactorData` = `PackedFunctionNodeData{InferenceType}`
`SolverData` -  Retrieve solver data structure stored in a variable/factor.
`VariableSolverData` - for get/add etc.

Related:
The observation function is currently part of the factor solver data, see https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/1119 and https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/992

dehann, 2025-03-26:
yes should be the same noun.  @Affie , please dictate us a noun.  We'll get to the actual work at some point.

dehann, 2025-03-26:
on use of `state` for either variable or factor, I find it useful to think about probabilistic chain rule:

the nav problem boils down to two sides of the same coin (think Bayes rule):
- sensors make measurements given the state of the world
- estimation inverts to find the state give the measurements

So, is a measurement a state?

I say yes because inside the sensor, the measurement process is resolving the best value (aka state) of whatever physics it uses.  So everything is actually just a state, but the topology of what influences what, and the causality about "chicken or egg first" is a seperate issue.  We approximate that in the bipartite graph (variables and factors).  i.e. we already know the TYPE of the the state from the bipartite nature of the graph.

I'm fine with state for both variables and factors.

Affie, 2025-03-26:
The implementation of the factor state part will be looked at in #992