JuliaRobotics/DistributedFactorGraphs.jl #1131: tag v0.26.0

dehann, 2025-05-06:
@JuliaRegistrator register branch=release/v0.26

dehann, 2025-05-06:
@JuliaRegistrator register branch=release/v0.26

Release notes:

## Breaking changes

- plot extension via different dependency, GraphPlot -> GraphMakie

JuliaRegistrator, 2025-05-06:
Registration pull request updated: [JuliaRegistries/General/130378](https://github.com/JuliaRegistries/General/pull/130378)

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.26.0 -m "<description of version>" eff25afe67f34f001128d09951e324dc45507696
git push origin v0.26.0
```

closed 2025-05-06 by dehann