JuliaRobotics/DistributedFactorGraphs.jl #1140: Restore for variables and factors the module prefix

dehann, 2025-06-23:
Restore for variables and factors the module prefix, e.g. "RoME.Pose2", or "Caesar.ScatterAlign".  Fallback is to search for the fields in `Main`.  FYI, this used to be in older DFG, probably just got lost in the shuffle. 

> BTW, fileDFG also does not have the module in factors, so also just Pose2Pose2. So, I would consider it a breaking change to to include it.  

 _Originally posted by @Affie i

dehann, 2025-06-23:
From Affie:

> Yes, that should definitely be there. If we run unique code without a package definition then we can either fall back to Main automatically, but first need to do a security audit on hte code. What if a user uploads a piece of malicious code and gets run through this evaluation. i think the security will be better by resitrcting to packages we know and trust as general rule, and then only deal with the Main exceptions.

Current deserialization calls eval which is very bad. I fixed it, but with new constraints I think we can live with.

Affie, 2025-11-10:
Here is an example of the leading idea to include the module information (for `Observation`):
```json
"observation": {
    "Za": {
      "μ": [
        1.0,
        2.0
      ],
      "Σ": [
        [
          0.1,
          0.0
        ],
        [
          0.0,
          0.2
        ]
      ],
      "type": {
        "pkg": "DistributedFactorGraphs",
        "name": "PackedMvNormal",
        "version": "0.28.0"
      }
    },
    "Zb": {
      "α": 2.0,
      "θ": 3.0,
      "type": {
        "pkg": "Distributions",
        "name": "Gamma",
        "version": "0.25.122"
      }
    },
    "type": {
      "pkg": "Main",
      "name": "TestObsAB",
      "version": null
    }
  },
```

If it ever becomes needed we can add `"pkg_id" : "[PKG_UUID]"`