JuliaRobotics/DistributedFactorGraphs.jl #1157: Make blobId=>Blob immutable

Affie, 2025-07-07:
According to [verb noun wiki](https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/wiki/Standard-Definitions-of-API-via-verbNoun-structure#noun-object), `Blob` is immutable. Currently, that is not enforceable because you can just delete the blob and then add a new blob with the same blobId. I would still like to be able to delete blobs.  Therefore, the pair blobId=>blob should become immutable.  Even if we make it immutable within a single blob store, it will still depend on the user to maintain the integrity across blob stores, perhaps using uuid7 will can also help.

Affie, 2025-07-18:
Sorry, I read through the last response a few times but I don't understand it. I also tried to find any design notes or discussions on blob immutability, but could only find a reference to some "Falmouth Manifesto". So perhaps part of this issue should be to document what decisions were previously made about this.

dehann, 2025-07-31:
From phone call, there was some miscommunication.  Punchlines are:
- We are not concerned with Blobentry mutations for this ticket (can discuss that elsewhere if needed)
- point of this ticket is to specifically about a blob being stored in different blob stores, and what blobId it receives.
- To enforce immutability as best as possible, the code API will only export functions that use labels as unique identifiers, while uuids are internal details for power users only.
- non-exported functions will be available for power users, but risks them overwriting blobs with same blobId in local in memory stores
- Blobstores implementation spec should include the "tombstoning" strategy whereby once an uuid is added, it cannot easily be removed -- hold an empty character, or `0xFF` or similar.
  - `FolderStore` will implement this pattern
  - Any managed service blobstores will be used in a "dumb" capacity to limit dependence on any particular provider, e.g. S3 has fancy tools for helping with tombstoning or un-delete, etc.  We're not interested.
- Future path for un-delete functionality will be achieved through a "trash/recyclebin" blobstore.

dehann, 2025-09-15:
right, sorry I was at the high level interface which mixed blobentry.label and blob. 

> there will also be a few high [level] helpers

I'm good with whatever you think is sensible here.  Guessing what a first time user would look for functions: `img_bytes = getBlob(var, :left_cam)` -- this is very high level as it assumes the search starts over blobentry labels attached to `var`, then works it's way through local cache, local, then remote blobstores, etc.