JuliaRobotics/DistributedFactorGraphs.jl #1172: Should Blobentry.metadata dict have CRUD?

dehann, 2025-09-16:
We have enough need for CRUD on agent.metadata.  Its a lot of work to add the same CRUD to blobentry.metadata, so likely can delay into DFG v1.X as feature expansion.

dehann, 2025-09-25:
just to confirm from the metadata design call,
- metadata field is purposefully kept as informal scratch space where user can dump any weird and wonderful data fragments they need to make their lives easier, with basically two conditions.  (See `Bloblet` for more formal counterpart to metadata),
  - don't supplant the use of blobs and blobentries for large chunks of data if you intend production use (obviously the development cycle of a few Mb landing in metadata is par for the course / it happens.
  - Whatever you store in any metadata field has one basic JSONable requirement which is enforced where the metadata field content has to be key=>String pairs (implemented via either vec or dict).
- the type for any node's `struct field metadata::Dict{Symbol,String}`
  - These key values can maybe go as far as atomic `add` verbs in DB case,
  - Non-power user has to overwrite the whole node to delete keys
  - Manipulating key values in-memory is a power user feature, which requires power user to know that updates are unsafe (non-atomic to Remote), unless the entire node is overwritten.
- Introduction of the high level convenience phantom noun `Bloblet` which will use blobless `Blobentry` to achieve non-poweruser CRUD/ACID for "bloblet" (aka small data) -- see new issue below

dehann, 2025-11-13:
What is the difference between String and JSONText?  

If it leads to escape characters then I'm hard no.

No crud on metadata.

Affie, 2025-11-13:
> What is the difference between String and JSONText?
> If it leads to escape characters then I'm hard no.

Not automatically escaped is the reason for the suggestion, so I take it this is resolved then.
`String`: Special characters (like quotes, newlines, etc.) are automatically escaped to produce a valid JSON string.
`JSONText`: The raw content is inserted into the output.